meshtastic/web #4: Unable to change channel presets

kingoffoxes, 2022-01-09:
When changing from the default "LongSlow" to any other setting the save icon at the bottom right remains disabled. 
Pressing "save" in the Channel 0 area doesn't commit these settings either. The same occurs when trying to set the settings without using the presets. an infinite loading symbol is displayed. 

When manually setting the modem config in python, the webpage doesn't reflect the settings. Changing from Bw125Cr48Sf4096 to Bw500Cr45Sf128.

sachaw, 2022-01-13:
@kingoffoxes Could you please confirm that the latest release fixes your issue, running the latest alpha you can visit `DEVICE_IP/admin/spiffs/update` to get the latest version.

kingoffoxes, 2022-01-13:
@sachaw So Ive updated to firmware-1.2.51.f9ff06b which has resolved the issue.... but the link shows a 404 not found at http://192.168.1.93/admin/spiffs/update and now i see we have just released 1.2.51 so i will try that.

closed 2022-01-14 by sachaw