meshtastic/web #6: missing mqtt settings

cyberlink1, 2022-01-14:
I noticed you can set a channel uplink and downlink but I cant seem to find a place to set the mqtt server address. 

I also noticed that while you can connect via ssl it just sits on the "Connecting" display and never connects to the unit. Connecting via http seems to work fine.

Firmware 1.2.52, full update of web interface.

sachaw, 2022-01-14:
Thanks for that, I'll add the mqtt details today, the https related issues, I may defer in favor of websockets, but only time will tell.

sachaw, 2022-01-14:
Please try out cdb11e106bcb961c93b374aa5ac628a402a539a1

cyberlink1, 2022-01-14:
Im not sure how to try it out, is there an upgrade button on the t-beam's web page?

sachaw, 2022-01-15:
If you are running the latest build, you can visit DEVICE_IP/admin/spiffs/update in your browser to update the ui

cyberlink1, 2022-01-15:
Im running meshtastic v1.2.52 and the web page it deploys. 

The link provided returned a 404, however going to <device ip>/admin then clicking on manage web page and clicking on the update there did work.

Thank you, and I did not even realize they just added mqtt username and password tell I saw it on the page. lol More fun stuff to play with ;)

closed 2022-01-15 by cyberlink1