meshtastic/web #11: Spiffs filesystem plugin delete button not working

cyberlink1, 2022-01-15:
I noticed this when I tried to delete the rangetest.csv clicking on the trashcan does nothing.

cyberlink1, 2022-01-17:
looks like the upload button is not working as well.

sachaw, 2022-02-17:
New filesystem browser is slowly being worked on, please update to the latest version.

closed 2022-02-17 by sachaw

pipedogtotherescue, 2022-02-20:
Does the latest tbeam firmware cover the filesystem browser UI?  (I assume so).  I am running fw 1.2.55.9db7c62 - just installed the latest a few days ago when I got my tbeam in the mail.
[image]

sachaw, 2022-02-21:
@pipedogtotherescue The trash can icon that's mentioned was from an older version from the UI,
As for not working in edge, could you please tell me what version of edge you are using, as there shouldn't be any issues.

pipedogtotherescue, 2022-02-22:
@sachaw ahh ok, good to know.  I am using Edge version 98.0.1108.56 -- it seems to work great.  The only thing that doesn't work is deleting rangetest, but seems to be a known bug.  Chrome browser doesn't work at all for me, but just noticed I am out of date and updating now.