meshtastic/web #7: Web interface comes up blank on firefox

caveman99, 2022-01-15:
up to 96.0.1 (which is current at the time of writing) the following error is logged:

```
Uncaught ReferenceError: ReadableStreamDefaultReader is not defined
    $I https://192.168.101.78/vendor.77bb4497.js:93
    <anonymous> https://192.168.101.78/index.cfc92925.js:2
vendor.77bb4497.js:93:16797
    $I https://192.168.101.78/vendor.77bb4497.js:93
    <anonym> https://192.168.101.78/index.cfc92925.js:2
    InnerModuleEvaluation self-hosted:2388
    evaluation self-hosted:2349
```

sachaw, 2022-01-15:
Hi, thanks for submitting this, I will need to file a bug report with Mozilla, as there documentation states that this part of the streams API has been implemented. The other strange thing is that the part of code that uses it should not be invoked on page load

caveman99, 2022-01-15:
not very savvy with ract, but this gets instantiated in the core components at https://github.com/meshtastic/meshtastic-web/blob/5b10d4de596a7e53222b1cb1f3c7f837c99d9676/src/core/connection.ts#L40 and then right again in Line 42 in setConnection. the first call might trigger this

sachaw, 2022-01-15:
Yeah, good point, although it should work anyway, so not gonna change that unless I have to.

s00500, 2022-01-27:
Safari on macOS also errors out on the same issue: ReferenceError: Can't find variable: ReadableStreamDefaultReader

If ISerialConnection is not used then why instantiate it? as far as I know chrome is the only browser that might provide some serial interface... no others do at the moment correct ?

sachaw, 2022-02-06:
This should now be "fixed" http works fine, webserial will still throw the same error.

closed 2022-02-06 by sachaw

5rxp, 2022-03-24:
@sachaw WebSerial is still not working on Firefox, is there a work around? I am getting the error:

`Uncaught ReferenceError: ReadableStreamDefaultReader is not defined`

sachaw, 2022-03-25:
Firefox does not plan on implementing the Web Serial API. So out of my control unfortunately. You'll have to use chrome for this functionality.

I'll end up adding a warning on the client.