meshtastic/web #17: Max length of text message is only 86 characters?

markbirss, 2022-02-07:


sachaw, 2022-02-17:
@markbirss You said this is also the case for the android app?

If that's the case, is this not a firmware issue? can you get it to display any characters via any other means: cli, ios etc?

markbirss, 2022-02-17:
@sachaw Same amount character using 

`meshtastic --port /dev/ttyACM0 --sendtext "123456789012345678901234567890123456789012345678901234567890123456789012345678901234567" --dest \!1cea1a76`

I send 87 only max of 86 appear

garthvh, 2022-02-17:
This must be a device display issue with the larger screens, I can send and receive 228 bytes from iOS to the webapi, moving back to device.

🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭🍭