meshtastic/web #23: Previous messages are lost after reconnect.

svtdoug, 2022-04-01:
Not sure is this is an issue or a limitation of the web interface.  For the Android app, if you disconnect or go out of BT range, then reconnect to device, previous messages are displayed.  For meshtastic.local, doing the same and there is no visibility for previous messages.  The thread is lost.  Is this something which can be overcome?  

Thanks,
Doug

IIMacGyverII, 2023-02-06:
I don't know how much can or should be stored in the meshtastic device storage but this sounds like something that could be easily handled by local storage in the browser.  This would mean that the browser would store the data and would have to be actively connected to the device to store the messages as they come through.  I believe the android/ios apps are doing basically the same thing.

ianmcorvidae, 2024-02-19:
This seems like it'd be a great application of [IndexedDB](https://developer.mozilla.org/en-US/docs/Web/API/IndexedDB_API) to me, potentially.

Hunter275, 2024-09-01:
Closing, more recent discussion: #162

closed 2024-09-01 by Hunter275