meshtastic/web #25: Settings menu does not reflect device settings, changes = lockup

charminULTRA, 2022-05-19:
When connected via Serial, the settings configured on the Meshtastic device (RAK 4631 in this case), are not accurately reflected in the web interface settings menu. For instance, I have set my radio to LongFast, US region, but the Lora settings on the Web interface says "VLongSlow" and "Unset".

Further, changing the setting and hitting Save results in the interface being blurred out, with "Loading" displayed indefinitely.

charminULTRA, 2022-06-18:
I can confirm this also happens on T-Echo, changing settings freezes both the Web interface and the device itself. Web interface just stays stuck on "Loading".

charminULTRA, 2022-06-18:
The changes also do not take effect.

sachaw, 2022-08-06:
This should now be working.

closed 2022-08-06 by sachaw