meshtastic/web #26: Web Interface Uses External Libraries - Can't access without internet?

accessiblepixel, 2022-06-02:
Hi,

I'm a new user of Meshtastic and I've just flashed my first two devices.

To get better access to configuration options I attempted to start the SoftAP on the device and connected my laptop to it. The interface comes up and is just white so I thought I'd check the html source code to see if it was actually getting html.

I've found the reason I'm having issue, obviously connecting directly to a Meshtastic device, you don't have any connection to the internet, and it seems that external requests for JavaScript libraries are required for any functionality on the web front end.

Is there a way to load those libraries onto the device to make it work? I'm unsure how it would otherwise work in a situation that you're offline (one of the primary reasons for the mesh network)?

Kind regards,
Jessica

garthvh, 2022-06-02:
The hope is to get more of the functionality on the device, currently there is not enough space for some fonts and maps.  Things like maps are probably always going to be too large.

garthvh, 2022-06-02:
It does not have offline functionality currently, pull requests for new features are welcome.

sachaw, 2022-06-03:
I'm aiming to have this fixed with the 1.3 rebuild so wait out for that

garthvh, 2022-06-03:
We need documentation and issues too.

sachaw, 2022-08-16:
@accessiblepixel this should no longer be the case with 1.3, it may re-occur in the future, if so please reopen this issue.

closed 2022-08-16 by sachaw