meshtastic/web #27: Cannot access the node web interface if GitHub API rate limit exceeded

fabfurnari, 2022-07-28:


mc-hamster, 2022-07-28:
moved to web

garthvh, 2022-07-28:
What firmware version are you running?

garthvh, 2022-07-28:
> IMHO a local web interface should be not dependent on external services to show.

I get this assumption, but it does not really match the state of mobile phone browser support or the amount of space available on the local web server.

The web ui is really used more as a base station / bridge where there is internet than as an offline mobile messaging client.  Cutting down on external dependancies is a goal, but there is a very limited amount of space. My guess is that these GitHub calls for releases are for doing firmware updates, there is no way to store all of these locally.

fabfurnari, 2022-07-28:
Hi, I'm using the `1.2.65.0adc5ce` with a LilyGo LORA V2.1 1.6.1

I understand the limited space on the device itself, maybe the error could be changed to a non-critical one?

sachaw, 2022-08-06:
@fabfurnari If you upgrade to 1.3 you should no longer have this issue, if that's not the case, please reopen this issue.

closed 2022-08-06 by sachaw