meshtastic/web #28: WiFi PSK validation does not allow for long keys

Noki, 2022-08-04:
A Wifi PSK can be 8 to 63 chars in ASCII / 64 chars in HEX. Long keys do not work:

[image]

The PSK validation should be adjusted.

caveman99, 2022-08-04:
Indeed the Protobufs have the right length

*WiFiConfig.ssid max_size:33
*WiFiConfig.psk max_size:64

garthvh, 2022-08-04:
The practical limit is in characters is going to be something less than the byte limit.  Probably about 60 characters for PSK, the text on the error is wrong, but the text on labels and validation are not finished, when I tested this allowed an expected password max length of about 60 characters.

garthvh, 2022-08-05:
63 characters saved which is the expected limit

Noki, 2022-08-05:
@garthvh can you please reference a commit that fixes the 16 char limit in the web client? Or why did you close this one?

garthvh, 2022-08-05:
We saved a 63 character value yesterday, but you are correct it was from the CLI, not the Web UI

sachaw, 2022-08-06:
Fixed in https://github.com/meshtastic/meshtastic-web/commit/258ffc2ec64640ff967e1b274ba6a79fc83f2076

closed 2022-08-06 by sachaw