meshtastic/web #29: Changing User-Config does not work on TBEAM

Noki, 2022-08-04:
Changing anything on the user config page results in the following error:

[image]

garthvh, 2022-08-04:
What browser are you using?  Both HWMODEL and MAC are disabled on that page for me.

Noki, 2022-08-04:
Latest version of Chrome:

[image]

Just try updating the Device Name and click save. That should reproduce it.

garthvh, 2022-08-05:
https://github.com/meshtastic/Meshtastic-device/issues/1590#issuecomment-1205315612

pcopa, 2022-08-05:
Ditto for me. Cant make a change on the user page due to the grayed out "TBEAM"

>  System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a20
 Executable: C:\Users\Schelto\AppData\Local\Programs\Python\Python310\Scripts\meshtastic
 Python: 3.10.5 CPython MSC v.1929 64 bit (AMD64)
 
Using Chrom latest version

Thanks for the good work

sachaw, 2022-08-06:
Fixed in https://github.com/meshtastic/meshtastic-web/commit/1dd239ca091ff411ac8f27915fc6c55f70389cc7

closed 2022-08-06 by sachaw