meshtastic/web #30: Web page never connects to either Serial or http

scarolan, 2022-08-12:
I have managed to install the web panel on one of my devices, but am unable to load info in the UI.  It just spins.  I get the same thing with your hosted version.

[image]

scarolan, 2022-08-12:
And this is what I see after connecting to Serial:

[image]

It also just spins like the example above.

sachaw, 2022-08-13:
Thanks for filing the issue, firstly the TBA-TBA you see under serial will be fixed soon, as for why it's not working is another story.

For http connections, there should be no issues whatsoever, could you please open your devtools F12 then try connecting again and tell me what the error is?

garthvh, 2022-08-13:
Based on an earlier issue I think he updated the webui from a 1.2.65 device is that correct @scarolan

sachaw, 2022-08-16:
@scarolan can you please confirm your firmware version?

sgofferj, 2022-08-18:
http does work locally but nor from client.meshtastic.org. That seems to be a mixed content issue. The web UI tries to contact http://meshtastic.local/hotspot-detect.html but client.meshtastic.org is served via https.

sgofferj, 2022-08-18:
@sachaw 
for serial I get a bunch of errors on the console:
```
[ERROR] iHttpConnection.pingFailed to execute 'fetch' on 'Window': The user aborted a request.
j_ @ index.a124fda7.js:97
log @ index.a124fda7.js:99
(anonymous) @ index.a124fda7.js:99
Promise.catch (async)
ping @ index.a124fda7.js:99
connect @ index.a124fda7.js:99
(anonymous) @ index.a124fda7.js:99
setTimeout (async)
connect @ index.a124fda7.js:99
await in connect (async)
(anonymous) @ index.a124fda7.js:99
```

rohanki, 2022-08-24:
> @sachaw http does work locally but not from client.meshtastic.org. That seems to be a mixed content issue. The web UI tries to contact http://meshtastic.local/hotspot-detect.html but client.meshtastic.org is served via https.

Same issue here. I have also tried using the IP address instead of meshtastic.local, and client.meshtastic.org will try and connect to my IP using http not https.

sachaw, 2022-08-24:
Can everyone please confirm they are running the latest 1.3.x firmware?

sgofferj, 2022-08-24:
1.3.38.
Mixed content errors have not much to do with the firmware though. That's an issue with the web client.

Noki, 2022-08-24:
As a workaround insecure content can be enabled for just a single site in chrome:

You can click the secure icon and go to "Site Settings". You can then allow "Insecure content". Afterwards you can access a meshtastic device in the network.

[image]

[image]

sachaw, 2022-12-03:
This should now be fixed as I force tls connections if the page is served over https

closed 2023-01-07 by sachaw

scarolan, 2023-06-01:
Rad thanks for the fix.