meshtastic/web #33: [Feature] Custom map WMS/TMS URLs

sgofferj, 2022-08-18:
It would be nice if it was possible to configure custom WMS/TMS URLs for the map of the web UI. That could either be in a config file on the web server or saved in the client browser.

Hunter275, 2024-06-16:
Description

Many people have requested the ability to change the map tiles or theme. https://github.com/meshtastic/web/issues/199#issuecomment-2154709194

Adding a configuration setting to allow a user to input their own .json map theme would allow for custom maps.

This would probably be a web only setting. The iOS app uses the built in map, unsure about the Android app.

skylarmt, 2025-07-08:
+1 to this.  I ended up editing the minified index JS file to set the map server to one I'm running locally, so the map will work without Internet access.  However, that would be much more annoying to do with the firmware-bundled web UI, I "installed" it on a PC.