meshtastic/web #40: Serial Disconnect/Reconnect Requires Browser Page Refresh

ghostop14, 2022-10-06:
With a serial-connected device (1.3.41 firmware), the web page in Chrome can connect to the device, but if you click disconnect then try to reconnect it will fail.  Workaround is to use the browser to reload the page, then the device will connect again.

Developer logs report this on disconnect:

[image]

Reconnect then reports "Failed to execute 'open' on 'SerialPort': The port is already open." until page is refreshed.

ajmcquilkin, 2023-01-08:
To give a bit more context on this issue, I can replicate this on Edge running firmware 2.0.12. I'm getting the following error when I disconnect from a serial device in the web client, and I'm not given the option to reconnect to the same device. Something to note is that this isn't necessarily a bug within the web client, since the error originates in the `iSerialConnection.ts` file in the [meshtastic/js](https://github.com/meshtastic/js) client library. The JS client is called in [src/components/Sidebar.tsx:34](https://github.com/meshtastic/web/blob/c0f7bc4741a25e9bb2369ddf929cb7ccc62db763/src/components/Sidebar.tsx#L33). It might be worth filing this in the other repo.

[image]

Hunter275, 2024-06-15:
This is probably a new bug but I cannot Disconnect through the Search menu

Hunter275, 2024-09-15:
Issue still exists after Search menu fix

closed 2025-03-22 by Hunter275