meshtastic/web #53: Client.meshtastic.org Channel Editor does not save uplink/downlink

arduionoGP, 2022-10-30:
Selecting Enable on Uplink and Downlink followed by Save does not save.
1.3.48, HeltecV2.1, MQTT, fixed GPS position enabled. Trying on default Primary channel.

arduionoGP, 2022-11-03:
For those of you who are here because you need these settings for MQTT to work the CLI commands from the command line are a bit verbose - these are for the default primary channel-->

meshtastic --ch-index 0 --ch-set uplink_enabled true
meshtastic --ch-index 0 --ch-set downlink_enabled true

Incidentally, there is a typo as to what Downlink Enabled does on the client.meshtastic.org Channel Editor pages also. See MQTT documentation.

ctandi, 2022-11-18:
I'm running into the same issue. However, I don't have that problem with every other channel. I created two custom channels and could enable up- and downlink just fine.

sachaw, 2022-12-11:
This should now be fixed in https://github.com/meshtastic/web/commit/05f72ec24c6d07229207f623d45fe48155e4a846
If you still have an issue, please let me know.
For any typos/issues you see fell free to open a PR!
Thanks.

closed 2022-12-11 by sachaw