meshtastic/web #56: BUG: Fix Position but not in the Web View

Dtrieb, 2022-10-31:
The Android App shows me my Position, but the Web means:

**Current Location
No Fix.**

[image]

Dtrieb, 2022-10-31:
Also didn't show my position in the map on the web.

sachaw, 2022-12-11:
Does your device have it's position listed on it's screen?

Dtrieb, 2022-12-14:
Now it works (2.0.7) but the location from the map jumps to Melbourne. If I switch between the entries it always get back to Melbourne.

GMart, 2022-12-16:
Similar issue for me. The map always starts at Melbourne. I am using a fixed location. I have tried clearing cookies, and no luck. Also, for me the Current Location sidebar says something like "Current Location

16T EL 96628 67738"
Which doesn't seem to be a Lat/long location. I think the formatting is bugged? 
The node I am connected to is running 
2.0.7.91ff7b9-d/20

The other devices on the mesh can see the location.

sachaw, 2022-12-16:
Currently the map doesn't auto center on your nodes (will do soon)
As for formatting, it's MGRS.
I'll make an option so you can pick the coordinate format of your choosing

sachaw, 2022-12-28:
This has now been implemented, When the map loads, it will do one of three things:
1. No nodes with position: Map center will be Melbourne as before
2. 1 node with a position: Map will center on that node with a zoom level of 12
2. multiple nodes: Map will create an automatic bounding box to fit all nodes within.

closed 2022-12-28 by sachaw

Dtrieb, 2023-02-12:
Reopen, because new Version (2707181) have the same problem.

sachaw, 2023-02-12:
Does any of your devices have a position? It should auto center on them.