meshtastic/web #58: Enhancement: Batch save for multiple setting changes or unloadable config

xman5735, 2022-11-08:
When setting up multiple meshtastic devices, taking the time for the device to reboot after every setting change takes extra time and seems redundant. It would be nice to make all my changes and then click save or to click save on each page and then reboot at the end.

Alternatively, being able to upload a preconfigured config file with the changes would work, but is less ideal in my opinion.

garthvh, 2022-11-08:
Each group of settings (config or module) causes a reboot in the firmware after they save this is to enable the use of the admin channel for management. The previous config got too large to send a n a mesh packet.

xman5735, 2022-11-08:
I see. It would be nice to commit the changes at the end of getting settings ready.  Then the webserver would run through the stack of changes one by one, including reboots, until all changes are applied. I understand that this would be a low priority QOL change. 
I appreciate the fast response.

sachaw, 2022-12-11:
This is now in progress, The UI is provisioned, just need to get it to work right.

sachaw, 2023-02-25:
Should work now

closed 2023-02-25 by sachaw