meshtastic/web #60: Bug: Force SSL redirect on client.meshtastic.org

D4rk4, 2022-11-21:
```
curl -i http://client.meshtastic.org/
HTTP/1.0 308 Permanent Redirect
Content-Type: text/plain
Location: https://client.meshtastic.org/
Refresh: 0;url=https://client.meshtastic.org/
server: Vercel

Redirecting to https://client.meshtastic.org/
```

For correct working with non-SSL meshtastic web API, redirect on the site must be disabled.

Why plain HTTP connection to a meshtastic device is better:
1) No MTU/MSS issues
2) Certificate on device regenerated with same serial -> no access from web after factory reset
3) WAN optimization working

So please give option to use HTTP://client.meshtastic.org

caveman99, 2022-11-21:
for self-hosted in an intranet the http option works. Browsers will not engage webserial or webBLE on remote http connections, and i think even for cross origin http access the TLS must be enabled. @sachaw is the expert on that. I stumbled across the serial number issue as well, this needs to be updated in the device firmware, to not pick the same serial again.

caveman99, 2022-11-21:
The certificate issue will be fixed in the next firmware with https://github.com/meshtastic/esp32_https_server/pull/3

sachaw, 2022-11-21:
I don't really want to support serving the page without TLS on the hosted version, you are free to self-host it with any webserver you want with TLS disabled

closed 2022-12-03 by sachaw