meshtastic/web #63: Page is not fully configured untill reconfigure

kingoffoxes, 2022-12-27:
TBeam - 2.0.8.090e166-d (had the same issue on 2.0.7.91ff7b9 so updated)

Connecting to TBeam via direct IP in browser (FF 108.0.1), selecting meshtastic.local without TLS connects to the node but there is no functionality and the node is "unknown" - see attachment.  (same issue with Edge 108.0.1462.54 )

Fix is to Ctrl-K and select Debug > Reconfigure. 
This works and i can see information and send messages. 

[image]

Incoming messages do not display until i send a message from the webpage but that may be an unrelated fault.

sachaw, 2022-12-28:
Can you please let me know what errors you are seeing in your console when it fails to configure?

kingoffoxes, 2022-12-28:
Seeing this in the console

[HTTPS:I] New connection. Socket FID=52
[HTTPS:I] Request: GET / (FID=52)
[ 27882][E][vfs_api.cpp:104] open(): /littlefs/static/index.html does not exist, no permits for creation
[HTTPS:I] Request: GET /index.ed76384f.js (FID=52)
[HTTPS:I] New connection. Socket FID=53
[HTTPS:I] Request: GET /index.c409c94b.css (FID=53)
[ 28084][E][vfs_api.cpp:104] open(): /littlefs/static/index.ed76384f.js does not exist, no permits for creation
[ 29617][E][vfs_api.cpp:104] open(): /littlefs/static/index.c409c94b.css does not exist, no permits for creation
[HTTPS:I] Client closed connection, FID=52
[HTTPS:I] Client closed (FID=52, cstate=2)
[HTTPS:I] Connection closed. Socket FID=52
[HTTPS:I] Client closed connection, FID=53
[HTTPS:I] Client closed (FID=53, cstate=2)
[HTTPS:I] Connection closed. Socket FID=53
[HTTPS:I] New connection. Socket FID=52
[HTTPS:I] Request: GET /Logo_Black.svg (FID=52)
[ 29794][E][vfs_api.cpp:104] open(): /littlefs/static/Logo_Black.svg does not exist, no permits for creation
snip - 
[HTTPS:I] New connection. Socket FID=53
[HTTPS:I] Request: GET /hotspot-detect.html (FID=53)
[ 37784][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html does not exist, no permits for creation
[ 37788][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html.gz does not exist, no permits for creation
snip -
[HTTPS:I] Request: GET /hotspot-detect.html (FID=53)
[ 47962][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html does not exist, no permits for creation
[ 47966][E][vfs_api.cpp:104] open(): /littlefs/static/hotspot-detect.html.gz does not exist, no permits for creation

Here is the debug from Firefox. i have marked where i manually hit Ctrl-K reconfigure to get it to load properly. 
[image]

sachaw, 2022-12-28:
Can you show me the console tab of your browser

sachaw, 2022-12-28:
Also, what are the response headers for the requests that had cors errors, they should not be appearing as per https://github.com/meshtastic/firmware/blob/master/src/mesh/http/ContentHandler.cpp#L684

kingoffoxes, 2023-01-01:
Here is the console. 
[image]

[ERROR] iHttpConnection.ping
❌ NetworkError when attempting to fetch resource. index.ed76384f.js:236:84
    E6 http://192.168.1.73/index.ed76384f.js:236
    log http://192.168.1.73/index.ed76384f.js:238
    ping http://192.168.1.73/index.ed76384f.js:238
    (Async: promise callback)
    ping http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    f http://192.168.1.73/index.ed76384f.js:252
    lt http://192.168.1.73/index.ed76384f.js:252
    D8 http://192.168.1.73/index.ed76384f.js:29
    O8 http://192.168.1.73/index.ed76384f.js:29
    R8 http://192.168.1.73/index.ed76384f.js:29
    u5 http://192.168.1.73/index.ed76384f.js:29
    _T http://192.168.1.73/index.ed76384f.js:29
    R1 http://192.168.1.73/index.ed76384f.js:29
    Kx http://192.168.1.73/index.ed76384f.js:32
    j3 http://192.168.1.73/index.ed76384f.js:29
    R1 http://192.168.1.73/index.ed76384f.js:29
    Ex http://192.168.1.73/index.ed76384f.js:29
    Y8 http://192.168.1.73/index.ed76384f.js:29
    xT http://192.168.1.73/index.ed76384f.js:29
    O1 http://192.168.1.73/index.ed76384f.js:29
    Wf http://192.168.1.73/index.ed76384f.js:29
    Wf http://192.168.1.73/index.ed76384f.js:29
    createRoot http://192.168.1.73/index.ed76384f.js:32
    <anonymous> http://192.168.1.73/index.ed76384f.js:839

and response headers
[image]

sachaw, 2023-01-01:
Can you please try the hosted version at https://client.meshtastic.org

kingoffoxes, 2023-01-01:
Looks like the same issue, 
Remote or Local / TLS enabled / Disabled.

[image]

sachaw, 2023-01-02:
That's a super outdated version, please clear your cache and try it again.

kingoffoxes, 2023-01-03:
Thanks clearing the cache improved the situation somewhat, that also fixed an issue where i couldn't see new incoming messages until i sent one. (only fixed for the online version - direct web connection to radio still does this)
So this does "work" but i have to manually enter the IP of the radio. 
i can ping and resolve meshtastic.local to the IP address on this PC so I'm not thinking its a DNS issue.  

[image]

[DEBUG] iHttpConnection.ping
Attempting device ping. [index.ed76384f.js:233:82](http://192.168.1.73/index.ed76384f.js)
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://meshtastic.local/hotspot-detect.html. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 200.

[ERROR] iHttpConnection.ping
❌ NetworkError when attempting to fetch resource. [index.ed76384f.js:236:84](http://192.168.1.73/index.ed76384f.js)
    E6 http://192.168.1.73/index.ed76384f.js:236
    log http://192.168.1.73/index.ed76384f.js:238
    ping http://192.168.1.73/index.ed76384f.js:238
    (Async: promise callback)
    ping http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238
    connect http://192.168.1.73/index.ed76384f.js:238
    (Async: setTimeout handler)
    connect http://192.168.1.73/index.ed76384f.js:238

Thanks

kingoffoxes, 2023-01-05:
Switched to my laptop with Chrome. Version 108.0.5359.125 (Official Build) (64-bit)
found that i couldn't connect to the radio at all via the client.meshtastic.org via DNS or IP. (Ctrl-K reconfigure had no effect)

[image]

If i point chrome address bar to the local IP directly and also manually enter the IP i am able to connect.
It seems to have cached the page from the web as its at a more recent version than whats on the ESP32.

sachaw, 2023-01-06:
Looks like a DNS issue, just confirming using the hosted version: https://client.meshtastic.org and entering your device ip directly, you are able to connect?
If that is the case, you may not have multicast dns on or multiple Meshtastic devices registering the same hostname:  `meshtastic.local`

kingoffoxes, 2023-01-09:
Thanks, 
Confirming i'm using https://client.meshtastic.org/
i am able to resolve my radio IP when i ping meshtastic.local in command prompt. 
When will the current online version of the web be placed into the firmware? may help in testing comparatively.

sachaw, 2023-01-09:
Could be that chrome isn't picking up your local dns servers, or there overwritten.
The latest webUI gets automatically bundled every time a new firmware is realeased.
I'll be adding PWA support to the webUI soon so people can lessen there reliance on the bundled version for offline use.

closed 2023-01-09 by sachaw