meshtastic/web #72: web client refresh issue on edge and chrome (map/device list)

nickjanzen, 2023-01-10:
both the map and device list do not seem to auto update, when the device gets an update, even though it is showing on the physical device. 

clicking around, or clicking the dots by the list will update, and then sometimes going to the map you see the positions jump around as they are processing the "queued" packets from the device.

sachaw, 2023-01-10:
Can you please confirm what version of the firmware you are running and where you are accessing the webui?

nickjanzen, 2023-01-10:
2.0.12.49172e41-d/20

https://client.meshtastic.org/

connecting with bluetooth

also would it be possible to show the callsign on the map instead of the box?

Logicbloke, 2023-05-26:
Probably a similar issue running 2.1.12, and client.meshtastic.org connect to T-Echo using Bluetooth Web API on Chrome on MacOS.

I can send messages directly or in a channel but don't really get them although I can see them getting displayed on the T-Echo. Then when I send another message through the channel or directly to a use, then it would show the message that should have been received.