meshtastic/web #86: Reopen, because new Version have the same problem.

Dtrieb, 2023-02-12:
Reopen, because new Version (2707181) have the same problem.

_Originally posted by @Dtrieb in https://github.com/meshtastic/web/issues/56#issuecomment-1427089576_

sachaw, 2023-02-16:
Do any of your nodes actually have a position fix?
as the map will center on your nodes, but only if they have a position

karamo, 2023-03-11:
In our Mesh (≈20 Nodes):
• All nodes have a position.
• some nodes with position will not display in the map.
• center to out nodes only with der dashed box top right.

micheljung, 2023-04-17:
My node also has a position and is visible on the map, however, it starts in Melbourne. Clicking the dashed box in the top right throws an error in the browser console:

```
dex-7e029ddc.js:703 Uncaught Error: coordinates must be an array of two or more positions
    at q$ (index-7e029ddc.js:703:35905)
    at s (index-7e029ddc.js:703:37560)
    at onClick (index-7e029ddc.js:703:38176)
    at Object.q9 (index-7e029ddc.js:37:9864)
    at Z9 (index-7e029ddc.js:37:10018)
    at K9 (index-7e029ddc.js:37:10075)
    at d2 (index-7e029ddc.js:37:31482)
    at NS (index-7e029ddc.js:37:31899)
    at index-7e029ddc.js:37:36812
    at ex (index-7e029ddc.js:40:36943)
```

commit 90fd35eb (2023-04-18): Fix camera repositioning for single node (#104)

* Fix camera repositioning for single node

Before, camera repositioning only worked when there were at least two nodes,
even though the code was meant to handle single nodes as well.

This is related to #86 but doesn't fix it.

* Center camera on nodes when opening map

Fixes #86
closed 2023-04-18 by sachaw