meshtastic/web #87: Missing Fields for (lng & lat) at Fixed Location

Dtrieb, 2023-02-13:
In the new Version the Fields for lng and lat at Fixed Location are missing.

garthvh, 2023-02-13:
Toggle fixed positon on and they show

Dtrieb, 2023-02-13:
> Toggle fixed positon on and they show

No in this version, not anymore.

dieterthiess, 2023-02-13:
[image]

sachaw, 2023-02-13:
Yeah, sorry, currently an omission, I will be adding a modal that lets you enter a fixed position, but I think a change in the protobuds is in order.

Hunter275, 2024-06-17:
Echoing what @sachaw this is going to take a protobuff change

OZ2CPU-thomas, 2024-11-17:
Why is this closed ?? when it is still not fixed ?
lot long fields DONT show up, when selecting fixed position.
any work arrounds ?

Hunter275, 2024-11-18:
> Why is this closed ?? when it is still not fixed ? lot long fields DONT show up, when selecting fixed position. any work arrounds ?

This has not been closed

Fredozen, 2025-02-13:
While waiting for the correction, is it possible to know the settings in the config.yaml file to activate the fixed position and the long/lat/H coordinates so that this is taken into account?

pleasantone, 2025-03-19:
The CLI command seems to be able to do this?

Fredozen, 2025-03-20:
Normally, yes, it should be possible. But you need to install the package

Fredozen, 2025-03-22:
I have a raspberry with a lora hat and obviously the meshtastic client can't communicate with it on an SPI.

Fredozen, 2025-03-22:
And in the config.yaml file I put:
location:
  lat: 49.xx
  lon: 6.xx
  alt: 210
  fixed: true
And yet I have the message:

[mqtt] MQTT Map report enabled, but no position available

Fredozen, 2025-03-22:
The only way I've found is to use the Android app to set the position, and it does it correctly.

twaddington, 2025-07-28:
I have started working on this in https://github.com/twaddington/meshtastic-web/tree/fix/87.

So far I've added the missing fields to the Position Settings page and am pulling in the current position.

From looking at the Android implementation a `setFixedPosition` function needs to be added to the `Device` interface. I'm still working that out.

I also need to fix the types for the Position form fields.

dzienisz, 2025-08-05:
Can I help you? It's madness it's not working for last 2/3 years.

dzienisz, 2025-08-05:
> I have started working on this in https://github.com/twaddington/meshtastic-web/tree/fix/87.
>
> So far I've added the missing fields to the Position Settings page and am pulling in the current [...]

Please make a draft PR

twaddington, 2025-08-05:
@dzienisz I'm chipping away at it. I should have a draft up this week.

NellowTCS, 2025-09-10:
> [@dzienisz](https://github.com/dzienisz) I'm chipping away at it. I should have a draft up this week.

Any updates on this?

dgottstein, 2025-11-16:
Hi,

Is there an update on this issue? I can‘t set fixed coordinates in any way.
I tried CLI via Serial (NoneType error), CLI via TCP (no error but position is not set), WebUI and iPhone App (no input fields available)…

Latest stable firmware, CLI and App-versions.
HW is Heltec Lora v3.2

Maybe there is at least a workaround?

Thanks

dzienisz, 2025-11-16:
I work here to do it #909. Please give me more time. It's almost done!

dzienisz, 2025-11-29:
Merged to test

danditomaso, 2026-10-17:
Fixed by #909 (6a7be99a): Position settings now have latitude, longitude and altitude fields for a fixed position, plus a coordinate picker. Closing, but please reopen if anything is still missing.

closed 2026-10-17 by danditomaso