meshtastic/web #91: deps install error

krasi-georgiev, 2023-02-17:
pnpm i
 ERR_PNPM_FETCH_404  GET https://registry.npmjs.org/@buf%2Fmeshtastic_protobufs.bufbuild_es: Not Found - 404

This error happened while installing the dependencies of @meshtastic/meshtasticjs@2.0.20-1

@buf/meshtastic_protobufs.bufbuild_es is not in the npm registry, or you have no permission to fetch it.

No authorization header was set for the request.
Progress: resolved 69, reused 67, downloaded 0, added 0

krasi-georgiev, 2023-02-17:
even trying to just add this dependency

pnpm add @meshtastic/meshtasticjs
 ERR_PNPM_FETCH_404  GET https://registry.npmjs.org/@buf%2Fmeshtastic_protobufs.bufbuild_es: Not Found - 404

This error happened while installing the dependencies of @meshtastic/meshtasticjs@2.0.20-1

@buf/meshtastic_protobufs.bufbuild_es is not in the npm registry, or you have no permission to fetch it.

krasi-georgiev, 2023-02-17:
commit 35c88b67dd951d6cd923bfa76b6a30c966324b8c

krasi-georgiev, 2023-02-17:
last working commit 270718148af4e09fd24f82dadf2ccf74e410c147

ajmcquilkin, 2023-02-17:
A working fix is to run the command `pnpm config set @buf:registry  https://buf.build/gen/npm/v1/`, although this doesn't solve the deeper problem. There's been a discussion around this in the Discord `#web` channel

krasi-georgiev, 2023-02-18:
ok thanks

closed 2023-02-18 by krasi-georgiev