meshtastic/web #97: Web flasher asks twice if you want to update or overwrite

bmidgley, 2023-03-05:
[image]

I appears the web flasher is asking a second time if you want to reset the settings. But it could be for some other purpose which is confusing.

caveman99, 2023-03-06:
we're using different flash layouts for erasing or updating. That's what you select with the dropdown. The second prompt is from the flasher itself and can not be disabled that easily, since it's 3rd party code. The Flasher in its currentform is only a makeshift solution till our own integrated flasher in the docs page is fully functional. This problem will be solved when we switch to the new code base.

pdxlocations, 2023-09-30:
> [image]
> 
> I appears the web flasher is asking a second time if you want to erase the settings. You choose once in the dropdown and again with this checkbox. But it could be for some other purpose which is confusing.

can this be closed?

closed 2024-01-15 by sachaw