meshtastic/web #111: Can't configure static IP config

micheljung, 2023-06-05:
If you try this:
[image]

It fails like this:
[image]
'invalid uint 32: string'` (for search purposes)

Because of this:
```JavaScript
/**
 * @generated from message meshtastic.Config.NetworkConfig.IpV4Config
 */
export declare class Config_NetworkConfig_IpV4Config extends Message<Config_NetworkConfig_IpV4Config> {
  /**
   *
   * Static IP address
   *
   * @generated from field: fixed32 ip = 1;
   */
  ip: number;

  /**
   *
   * Static gateway address
   *
   * @generated from field: fixed32 gateway = 2;
   */
  gateway: number;

  /**
   *
   * Static subnet mask
   *
   * @generated from field: fixed32 subnet = 3;
   */
  subnet: number;

  /**
   *
   * Static DNS server address
   *
   * @generated from field: fixed32 dns = 4;
   */
  dns: number;
```

aptonline, 2024-02-12:
Seeing the same issue, unable to configure static ip settings under network. Hitting save does nothing.

aptonline, 2024-02-13:
I've seen over posts saying you have to enter static details in reverse for this to work (1.1.168.192 for example) .  Not tested but will be trying it.

aptonline, 2024-02-16:
That doesn't work for me, still not able to say settings for network (static IP etc) from WebUI. updated to latest beta firmware and same issue.

aptonline, 2024-02-18:
Managed to set static IP via Android app but noticed that the IP details are encoded, is this correct?  See below.

[image]

KomelT, 2024-09-17:
@Hunter275 this was fixed in #269.

closed 2024-09-17 by Hunter275