meshtastic/web #124: [Bug]: Nothing happens after clicking ‘Connect’ on the device-served website

dm5tt, 2023-08-11:
### Category

Other

### Hardware

T-Beam V1.1

Tested with 433 and 868 variant

### Firmware Version

v2.2.0.9f6584b

### Description

I just Web-flashed the latest 2.2.0.9f6584b Alpha onto my T-Beam (433 and 868MHz variant). After pressing the ‘Connect’ button on the website served by the device nothing happens.

A full device erase was done. Also cleaning out all browser caches. Browser logs (Developer mode) showing no error/warning.

Last known working version was V2.1.21 or V2.1.20.

Just tested the Android App against the IP - connects without problems.

Is there some known issue?

Thank you!

dm5tt, 2023-08-12:
I bisect'ed this twice and alway endet up here:

ajmcquilkin, 2023-08-13:
Just to clarify, does this issue happen on one connection method specifically? I'm on protos `v2.0.0 alpha` and I'm seeing this when trying to connect to a TBeam over USB Serial

[image]

daniel-hofer, 2023-08-16:
> > I only tested using the IP.
> > On the forum this issues now way confirmed this.
> > One person suggested a workaround: Marking the IP-Field then hitting enter. Works for me.
> 
> I can confirm this as well.

I can confirm this too, the workaround did it for me.

closed 2024-09-15 by Hunter275