meshtastic/web #129: [Bug]: Firmware 2.1.23 - Peer Messaging - Error received for packet #########: NO_CHANNEL

prurigro, 2023-09-01:
Using the web ui, I'm able to receive direct messages, and can both send and receive messages in the Primary channel, but if I try to send a message directly to a peer it immediately shows a circle with an exclamation point, and in the console log it says `Error received for packet #########: NO_CHANNEL`.

There doesn't seem to be a way to configure the channel that gets used when sending a message to a peer, but the fact that they're peered and the node can receive messages makes me think this is likely a bug?

This is sending from the Station G1 with firmware 2.1.23- the other device is on the same firmware and is communicating through the android app. Let me know if there's any specific information or logging that would help. Unfortunately the station can't both be powered enough to send messages and connected via serial at the same time, and bluetooth to the webui doesn't seem to work (it connects and appears to populate, but all the fields are empty).

Thanks!

garthvh, 2023-09-02:
Direct messages are always sent on the primary channel. That error is usually the result of a channel configuration problem, routing errors are not usually a bug.

prurigro, 2023-09-03:
Hmm, all I did after wiping was configure the country, long slow and name on both devices, and enable wifi on the router. Is it possible there's an issue with the defaults?

I'll try a few more configurations to rule things out once the long weekend is over.

Thanks for your time!

pdxlocations, 2023-09-04:
I tried loading older js dependancies and narrowed it down to V2.1.20-1
There's an issue with https://github.com/meshtastic/js/commit/9a58223740aafd198da953cd306230cb5cd23761
It's causing other problems such as:  #125 

Likely Line 119 of src/types.ts:

https://github.com/meshtastic/js/commit/9a58223740aafd198da953cd306230cb5cd23761#diff-c54113cf61ec99691748a3890bfbeb00e10efb3f0a76f03a0fd9ec49072e410aR119

export enum ChannelNumber {
  PRIMARY = 1,
  CHANNEL1 = 2,
  CHANNEL2 = 3,
  CHANNEL3 = 4,
  CHANNEL4 = 5,
  CHANNEL5 = 6,
  CHANNEL6 = 7,
  ADMIN = 8,
  }

I think PRIMARY is mapping to channel 1 instead of channel 0

pdxlocations, 2023-09-04:
I'm pretty sure this isn't firmware, it's web

prurigro, 2023-09-05:
@pdxlocations Looks like you're right- I just changed `PRIMARY = 1` to `PRIMARY = 0` and messages started sending correctly!

closed 2023-09-07 by thebentern