meshtastic/web #144: Easy install, and run?

Miaoucat0, 2023-11-20:
hello.

normal person here with (almost) zero programation experience. 
i am wondering if it would be possible to compile? or make it into a simple "click and go" type of software?
currently it seems the user need to go into command line, install a couple of programming language and stuff, then try to build the software and start it. i didnt even manage to run this myself...

i think being able to use a PC as a meshtastic handler/client is very important.
people are spending thousands of hours designing standalone meshtastic nodes, laptop PCI cards, and modules. 
being potentially able to use just about any computer, connect to a node and use it would be very good.

it would be great if the user installation would be simplified.

Thank you for your time.

OishiiCha, 2023-11-22:
Hi, the easiest way to do it would be via Docker.

First install docker (docker desktop for mac and windows)

Then run the following for the current version to date.
```
docker run -p 8080:8080 -p 8443:8443 --restart always --name Meshtastic-Web ghcr.io/meshtastic/web:8288b8c4fb42386b9e072773dc05a429b4527e58
```

Hopefully that helps.

Miaoucat0, 2023-11-22:
@OishiiCha  Thank you very much for your help! i will try this in the following days.
basically it make a "backup" of the site to run locally?

OishiiCha, 2023-11-22:
@Miaoucat0 The devs have precompiled a docker version of the current site so if there are any updates to the site that you would like, you will have to run a new docker container.

Miaoucat0, 2023-11-22:
@OishiiCha  alright! thank you very much for your help.
i havent tried it yet, but is docker able to run the site Backup offline after the initial setup? that would be my main goal.

OishiiCha, 2023-11-22:
Yes it will run without the use of internet, however I am yet to get the Bluetooth and serial to work, however Lan connections should work

OishiiCha, 2023-11-22:
Just an update, was able to get serial and bluetooth to work, there are different versions ARM or X86: 
[image]

Miaoucat0, 2023-11-22:
@OishiiCha  OK! thank you for much for your help!

marek22k, 2023-11-23:
How can you save the chat history (if this is even possible) with Docker?

sachaw, 2023-12-17:
Instructions now in readme

closed 2023-12-17 by sachaw