meshtastic/web #146: Peers list - sorting not working

stylesuxx, 2023-12-20:
The peers list table has carets in the column headings, indicating that it should be possible to sort by those fields, which does not seem to be the case. It would be nice if we were able to at least sort by `name` and `last heared`.

tekstrand, 2024-01-29:
@sachaw I came across this the other day and wanted to check with you on using `react-table` or something similar to solve. Any thoughts? https://www.npmjs.com/package/react-table

sachaw, 2024-01-29:
We're planning some pretty big changes, still a ways off though. Feel free to open any PR's

tekstrand, 2024-01-29:
@sachaw where is that planning happening? Would enjoy being part of that.

sachaw, 2024-01-29:
Still in the planning phase. We're looking to use a WASM plugin system. With plugins likely written in rust. Frontend will likely move to SolidJS. If you're interested, DM me on discord.

PR #165 (2024-06-21): Fix sorting on peers table
The peers list table has carets in the column headings, indicating that it should be possible to sort by those fields, which does not seem to be the case.

The PR enables sorting by clicking column heading, and defaults to most recent contacts at the top

Closes #146
closed 2024-06-21 by Hunter275