meshtastic/web #147: GPL 3.0 vs CLA

stylesuxx, 2023-12-23:
Could someone please give me a rundown of how the GPL v3.0 section about patents:

> Finally, every program is threatened constantly by software patents.
States should not allow patents to restrict development and use of
software on general-purpose computers, but in those that do, we wish to
avoid the special danger that patents applied to a free program could
make it effectively proprietary.  To prevent this, the GPL assures that
patents cannot be used to render the program non-free.

holds up to section 3 of the [CLA](https://cla-assistant.io/meshtastic/web).

Or to boil it down to one question: Will my contributions be potentially used in any kind of software patents?

garthvh, 2023-12-25:
They are for different purposes, not sure exactly why you think they are in conflict, does not read as a conflict to me.

sachaw, 2023-12-25:
Meshtastic does not hold any software patents.
The restriction under GPL is that any software (usually dependencies) of a GPL licensed program can only be GPL licensed if it remains free regardless of any patents.

mc-hamster, 2023-12-25:
Section 3 of our CLA protects Meshtastic in the event a contributor who owns patents, submits code that applies those patents, and then decides to turn around and get royalties or sue because of alleged violations of those patents.

stylesuxx, 2023-12-25:
Thanks @mc-hamster for taking the time to explain in depth, I understand it now.

mc-hamster, 2023-12-25:
@stylesuxx Anytime mate!

closed 2023-12-25 by stylesuxx