meshtastic/web #150: Encrypted messages are sometimes ignored

dustin, 2024-01-08:
I've been trying to figure out the pattern here, but it's not clear.

I was originally using Safari and things weren't working, so I tried Chrome and could see responses.  I then tried Firefox and that also failed.  Then Firefox started working and Chrome started failing.

I see this message in logs:

```
handleMeshPacket 🔐 Device received encrypted data packet, ignoring
```

With two browsers connected at the same time, I got a message in one but not the other.  (though I wasn't getting any messages in Safari with no others connected).

garthvh, 2024-01-08:
Is this a single device?

pdxlocations, 2024-01-08:
> I was originally using Safari and things weren't working, so I tried Chrome and could see responses. I then tried Firefox and that also failed. Then Firefox started working and Chrome started failing.

Compatibility table from  https://meshtastic.org/docs/software/web-client#compatibility
[image]

dustin, 2024-01-08:
I should clarify, this is using the web API.

I have one device on the network.  I don't understand the architecture very well, but it appears that sometimes messages are decrypted before being delivered to the browser and sometimes they're not (based on the error message I got when the message isn't displayed).

I don't know that it ever worked in Safari.  It works sometimes in Firefox.  It seems to work often in Chrome, but sometimes doesn't.  I think it's the same error and I might just be getting lucky.

I also have it delivering the messages unencrypted to my mqtt broker.

garthvh, 2024-01-08:
One device supports one client  connection, safari is not supported

dustin, 2024-01-08:
OK.  So one of Firefox or Chrome should be OK, but not two.  And Zero of Safari.  Makes sense.  Thanks.

closed 2024-01-08 by dustin