meshtastic/web #159: Node Map using web browser cannot zoom in and keeps resizing to node boundries.

TechMindsYT, 2024-02-12:
Hi,
I am accessing my node using Wifi and a chrome browser via the IP address.

When I select MAP, the map will take a while to move to the location of the discovered nodes, but once there I cannot zoom in and out of the map. Sometimes the node names do not show. If I move off the area where the nodes are and try to zoom, the map auto moves back to the nodes.  Please see the video:

[](https://youtu.be/VppJNuUmtyg)

Anyone else see this issue? Is there a fix?
Thanks,
Matt

pdxlocations, 2024-02-13:
> Is there a fix? 

Yes!  You'll need to update your node with a full erase and install to update the embedded web client.  A regular firmware update doesn't update the web client. You know you're up to date by checking for the latest commit hash in the bottom left:

[image]

TechMindsYT, 2024-02-17:
Thanks, this is fixed with latest firmware.

closed 2024-06-15 by Hunter275