meshtastic/web #162: Chat History on web client

imzenreally, 2024-02-17:
Chats doesn't retain history on the web client - I'd think designating a SQLite or something to live on a as card would be easy but I'm no dev

Is this being worked on?

TechMindsYT, 2024-02-17:
I second this.. I use web gui for my home install thats on wifi. Would be nice to have the messages retained somehow.

wojtasie, 2024-02-17:
+1 on this

ianmcorvidae, 2024-02-19:
Looks to be the same issue as #23 

(I agree, for the record. either IndexedDB or LocalStorage might be usable for this)

bmatic385, 2024-03-11:
+1 on this

ayysasha, 2024-03-15:
mega +1

GiovanniMoretti, 2024-05-20:
+1 Also. 

A related issue is that there's no "recent messages" tab on the web client. On Android you can see the most recent message in the Messages (leftmost) Tab but on the web client you have to click on each node to see if there's any new messages from that sender. This is impractical with dozens of visible nodes.

Having a Chat History tab where clicking on a message shortname takes you too that conversation would really make the web client usable as a day-to-day client.

wesker-albert, 2024-08-31:
This seems like a no-brainer, and would take the web client from a novelty to regularly usable.

Hunter275, 2024-09-01:
The web client currently does not have any way to store messages and the devices do not have enough memory to keep a message history.

We are looking into possible client-side storage solutions.

andrewheadricke, 2024-11-05:
As others have said, IndexedDb or LocalStorage would be a relatively quick and easy solution to get started. Would probably have enough storage to keep all but the longest of conversations.

I started messing around with a PoC the other day, it should be pretty straight forward although I did run out of time.

andrewheadricke, 2024-11-11:
Here's a basic chat history implementation using localStorage. Plus unread notification icons. Could do with some tidying up but it basically works.

https://github.com/meshtastic/web/pull/342

blaknite, 2024-11-14:
I've been inspired by @andrewheadricke's work and fleshed out a localStorage solution in #346

ToeiRei, 2024-11-15:
I am starting to think that in the end something like a 'proxy' in the middle would make the most sense for static stations - something where a client can pull chat history from.

andrewheadricke, 2024-11-16:
Liam Cottle built something similar to that but for all radio packets not just messages. https://github.com/liamcottle/meshtastic-websocket-proxy

Adding a middleware process would work well for the more tech savvy, but for a quick and easy web UI I think the localStorage is a good compromise. I've been using it daily for the past week, works great.

danditomaso, 2025-03-29:
So this feature has been completed but is still being reviewed in PR #536. I will update this issue when the code has been merged,

PR #536 (2025-04-04): Add message persistence using IndexedDB

## Description

This PR adds message persistance backed by the Zustand storage interface and IndexDB. 

Additionally, this PR ensures messages are structured to support efficient persistence in IndexedDB, enabling long-term storage of message history.

Because of the  refactor of how messages are stored this will require a good deal of testing prior to merging it. 

## Related Issues

I cant find the issue to link this to, but I know its been requested in the past
Fixes #162 and many, many other user requests on discord.

## Changes Made

- Extracted me [... cut]
closed 2025-04-04 by danditomaso