meshtastic/web #167: Enhancing Meshtastic Accessibility: A Call to Support the Heltec V2 firmware through the webflasher.

5kamisama5, 2024-02-24:
My thought will always be to be as beginner-friendly as possible. I've noticed that many newcomers, with limited technical knowledge, discover Meshtastic through social media and other channels. In their initial research, some end up purchasing the Heltec V2 by mistake. While it's not the ideal device for Meshtastic, as many like to point out, I believe in supporting those who have shown interest and taken the first step by investing in a board. Telling them they've got the wrong thing or that it's 'not supported' can be discouraging and may dampen the enthusiasm of someone who could become a valuable contributor to our project in the future.

Asking these beginners to navigate the less user-friendly CLI to install firmware can be daunting. To make their journey smoother and more welcoming, I kindly request the addition of the V2 to the web flasher. This small gesture can make a significant difference in their initial experience and encourage them to continue exploring the world of Meshtastic, which I've come to love.

I understand that this request might not align perfectly with the current roadmap, but I hope the developers can consider the needs of our non-technical users. Their early experiences can shape the future of our community, and a little flexibility now could lead to a more inclusive and thriving Meshtastic in the future.

[image]

garthvh, 2024-02-24:
It was removed completely, it is not built by the CLI anymore either.

5kamisama5, 2024-02-24:
Those who accidentally purchase the V2 are usually beginners, not experienced enough to know what to look for. Telling a novice to either throw it away and spend more to replace it or learn to use the CLI seems a bit shortsighted and feels like gatekeeping. Plus, a quick search on Amazon shows the V2 still prominently listed. If it's still widely available, shouldn't we consider supporting it to make things easier for newcomers? Unless I'm mistaken about what our goal here is. If so I apologize. @garthvh 

[image]

Sorry about putting this on the wrong place, not sure how to move as I am not an advance user myself.

closed 2024-02-25 by 5kamisama5