meshtastic/web #169: [Bug]: GPS issues

Slyke, 2024-03-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.23.5672e68

### Description

On the Web configuration, I can see these 3 options for GPS:
* Fixed Position
* GPS Enabled
* GPS Mode

On the Android App, I only see 2:
* Use fixed position
* GPS Mode

When ticking `Fixed Position` on the Android App, 3 textboxes show up: Latitude, Longitude, and Altitude. This does not show up in the Web configuration.

When ticking `GPS Enabled` in the Web configuration and saving, the device reboots, but the option stays off, it doesn't seem to be doing anything.

If I use the Android App to set a fixed GPS location, it doesn't seem to work unless I change the position from `0, 0`. The fixed position also isn't sent in the MQTT payload (at least the readable JSON version). Using the Web configuration to set a fixed position doesn't do anything at all, unless I close it and load up the Android App and set the latitude and longitude.

My physical GPS cannot get a lock, and I'll need to test it on the weekend when I have some time. I'm not sure what I need to do to get the Android App to use my phone's GPS.

It says "All position config options are available in the Web UI.", but I don't see a way to set a fixed position ( https://meshtastic.org/docs/configuration/radio/position/#position-config-client-availability ).
[image]

I'm still learning Meshtastic, so this was a little confusing.

jp-bennett, 2024-03-07:
It'd be nice to have the position_precision setting available, too.

kittenarmy, 2024-03-15:
I have the same issue with Heltec v3. I tried connecting to a Heltec v3 I had already configured a fixed GPS on Android to see if the fields would appear but they did not.

Hunter275, 2024-06-15:
Duplicate of #87

Position Precision was added recently though

closed 2024-06-15 by Hunter275