meshtastic/web #176: [Bug]: Saving MQTT config via web interface on Heltec v3 doesn't work

Calimerorulez, 2024-03-21:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.1

### Description

Trying to save MQTT config via web interface doesn't work.

### Relevant log output

```Shell
Console output Chrome:

index-nsbd6Oaw.js:84 Uncaught (in promise) TypeError: o.toBinary is not a function
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
    at Object.writeMessage (index-nsbd6Oaw.js:84:11921)
    at i.toBinary (index-nsbd6Oaw.js:83:29735)
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
    at Object.writeMessage (index-nsbd6Oaw.js:84:11921)
    at i.toBinary (index-nsbd6Oaw.js:83:29735)
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
```

yaleman, 2024-05-02:
I'm getting the same issue on T-Beam 2.3.6.7a3570a Beta ..

kdkd, 2024-05-09:
I can confirm the same problem. I think commit 6113924e462fabf63a2e9f8b841add0aaa4362fe broke it. Rolling it back allows me to enable MQTT now.

Obliterous, 2024-06-19:
I can confirm this is still broken with latest version as of June 19th.

Hunter275, 2024-06-19:
@Obliterous What firmware are you on and are you using https://client.meshtastic.org ?

Obliterous, 2024-06-19:
> @Obliterous What firmware are you on and are you using https://client.meshtastic.org ?

2.3.10.d19607b

Yes.

Hunter275, 2024-06-19:
@Obliterous Can you update to 2.3.11 and lmk if this bug persists, I cannot replicate it on 2.3.11

Obliterous, 2024-06-20:
2.3.11.2740a56 Beta installed, still unable to save mqtt config.  if I change & save any other setting for any other module, it will save those changes again without retaining MQTT changes.

Obliterous, 2024-06-20:
update: it always triggers a save of SERIAL settings, even if others have been changed. Changes to MQTT settings are not retained.

Obliterous, 2024-06-20:
Additional; after updating to 2.3.11.2740a56, I am no longer able to connect via BT.

dmanierre, 2024-06-21:
I'm also facing the same issue, I tried with 2.3.11 and 2.3.12 I was just trying to toggle the Enable slider on the MQTT tab and save but can't get the changes to stick.