meshtastic/web #178: [Bug]: webgui shows too many devices

cracky22, 2024-03-26:
### Category

Other

### Hardware

Heltec v2.1, Heltec V3

### Firmware Version

2.2.24, 2.3.1 and 2.3.2

### Description

https://github.com/meshtastic/firmware/assets/104385850/8b677067-3a11-48cd-acd8-aa7106aec7c4

I don't know if it's a mistake, but in the device list (peers) of my repeaters you can see a suspiciously large number of devices. Is this a bug or are these certainly Meshtastic devices? Could it also be that there are IoT devices nearby that also use Lorawan?

birdpump, 2024-04-16:
Also having the same issue, Android client shows known peers nearby but web interface shows a ton of random unknown peers along with the known nodes. Both Android and web were connected to my meshtastic device with serial.

GUVWAF, 2024-05-04:
Android doesn't show nodes it didn't receive a NodeInfo packet from (which include their name), while the Web UI does.

closed 2025-03-22 by Hunter275