meshtastic/web #179: Feature Request: traceroute for webgui

cracky22, 2024-03-30:
The option to traceroute in the peer list would be great :)

cracky22, 2024-05-11:
@caveman99 @geeksville What do you think about the idea?

sachaw, 2024-05-12:
I agree, it is certainly a useful feature to have in the web UI. Feel free to have a go at implementing it, I provide guidance if required.

cracky22, 2024-05-13:
> I agree, it is certainly a useful feature to have in the web UI. Feel free to have a go at implementing it, I provide guidance if required.

Hello Sachaw, I just had to smile a little :) I am a student (16 years old) and am graduating from secondary school this year (I come from Germany, Munich) but I would like to try to incorporate it when I get the chance / time. Do I need certain knowledge? I could offer html/js/css as well as python :)
thank you also for your support

sachaw, 2024-05-15:
That's great. Prerequisite knowledge isn't strictly require. Just some familiarity with JavaScript should suffice.
It's work looking through the codebase to wrap your head around what's going on. If you make any changes, feel free to open a PR and I can give feedback/assistance as required

fifieldt, 2024-05-26:
Hi @sachaw , @cracky22 -- I've hacked together the start of some basic functionality over here: 

https://github.com/meshtastic/web/pull/211 

I probably won't be able to progress that much more before I go on holiday, so if anyone wants to take it over and hack on it over the next couple weeks please do feel free!

fifieldt, 2024-06-15:
I've been using the code in #211 for the past couple weeks and asked some people in the local community to test as well. Result is: it works reliably :)

I don't have the skill to design a great UI integration, so hoping someone might be able to help get this over the line.

closed 2024-07-04 by Hunter275