meshtastic/web #186: Unable to connect to device using the web client

lazarvgd, 2024-04-26:
Hi, I have Heltec v3 and Tbeam devices, both flashed with the latest firmware **2.3.6.7a3570a Beta** and I cannot connect to them using Chromium based browser, I use online hosted web client https://client.meshtastic.org/.

I tried bluetooth and wifi on TBeam did not work. For heltech I tried, bluetooth only, and the same issue is present. The client kinda connects to the device but web says Loading... I cannot see channels, nor can configure the device, the web interface is clickable.
I tried Brave, Chrome, Chromium, Thorium browsers and it is the same problem.

Console of the page shows this
[image]

The device can be access via android/ios, no matter of the interface(bt, wifi) and I can ping it(wifi).

As a matter of fact, both devices work just fine with Android and iOS apps, bluetooth or wifi.
I am attaching image of the web page.

### I am not sure if I am doing something wrong or this is a bug?  Can you please advice?

[image]

gala-vine, 2024-04-28:
I have issues with my T beam also with the web client.. connecting through wifi.. lets me connect but then it will either let me send one message or none at all with  '2.3.6.7a3570a Beta' ? ?. can't get to it currently to check debug log through serial.

Talie5in, 2024-05-16:
Is this due to the issues with Firmware 2.3.7 that was Revoked? (and other issues in 2.3.8)
Does the client work for you if on 2.3.6 or 2.3.9 Firmware Release?

https://github.com/meshtastic/firmware/releases

lazarvgd, 2024-05-16:
I just flashed 2.3.6.7a3570a beta firmware and similar thing is happening.

To clarify, I have 3 devices, two heltec and one tbeam device. All devices can be accessed via bt, Tbeam does not work with wifi, I am unable to connect.
And Heltech devices cannot be connected via serial comms and I have ensure that are not connected via bt.

gala-vine, 2024-05-16:
> Are you able to give Edge a go? You will need to open the IP of your device first to accept the self signed cert.

No does the same in Edge as well, i'm using Opera.. i've read somewhere that only a erase/flash resets the web client on the device and not just an update without a flash.. i need to try this next, unless someone tries it before me.

gala-vine, 2024-05-16:
ok so a Erase then flash fixed my issue.

closed 2024-06-15 by Hunter275

Sawadee2u, 2025-01-28:
Having the same issue (Can't connect to unit {LiliyGo T-Beam Supreme V3.0) using IP address. Was working when first programmed but after running for three days and trying to connect again using IP address, can't connect. 
Have IP address being assigned by router and see it on the last screen of device. 
Also connected to on-line map using MQTT (Green dot on https://meshtastic.liamcottle.net/). 
Have tried Win11 edge and Chrome.
Have reprogrammed with and without erase with Mashtastic flasher without improvement. 
Using Mashtastic Ver.2.5.18.
Not sure what else to try.

Hunter275, 2025-01-28:
> Having the same issue (Can't connect to unit {LiliyGo T-Beam Supreme V3.0) using IP address. Was working when first programmed but after running for three days and trying to connect again using IP [...]

Please create a new issue, it will be easier to track that way