meshtastic/web #188: Unable to pull container with Docker

RobFisher, 2024-05-01:
I have this error when running the docker command (though I am doing it via Portainer, not directly):

`11.6MB/17.63MB 65aa6b7fe101 Extracting [==================================================&gt;]
17.63MB/17.63MB failed to register layer: lsetxattr user.overlay.impure /etc: operation not supported`

Reading around, this seems to affect several projects' docker images. It looks to be related to this limitation in recent Docker versions: https://docs.docker.com/engine/release-notes/25.0/#known-limitations

Here is an issue on another project dealing with the same thing, where they are planning to rebuild with updated buildah:
https://github.com/curl/curl-container/issues/55

It's possible that this is caused by the upstream nginx image, in which case perhaps updating that might help.

PixnBits, 2024-05-18:
Can you try deploying a sample using just the base image (`registry.access.redhat.com/ubi9/nginx-122:1-45`)? :thinking: 
Might be useful to also try [the official nginx image](https://hub.docker.com/_/nginx) `registry.hub.docker.com/library/nginx` directly, and then (maybe :sweat_smile:) an image built from this repository but substituting the latter for the former (that is, `FROM registry.hub.docker.com/library/nginx` https://github.com/meshtastic/web/blob/master/Containerfile#L1) :crossed_fingers:

danditomaso, 2025-02-28:
@RobFisher Please follow up if this issue is still effecting you. Otherwise this issue will be closed due to inactivity.

RobFisher, 2025-03-02:
> [@RobFisher](https://github.com/RobFisher) Please follow up if this issue is still effecting you. Otherwise this issue will be closed due to inactivity.

I just tried and I was able to deploy the container with no problem. Many thanks.

closed 2025-03-02 by RobFisher