meshtastic/web #199: [UX] Dark mode should change map into Dark Mode as well

Talie5in, 2024-05-16:
Dark mode setting should change map into Dark Mode as well, the bright map can cause eye strain when working in darker environments.

Hunter275, 2024-06-07:
The map is created using [react-map-gl](https://github.com/visgl/react-map-gl) as a wrapper for [Mapbox GL JS](https://github.com/mapbox/mapbox-gl-js). 

The map images themselves come from OpenStreetMap Carto as a raster as defined [in this json file](https://raw.githubusercontent.com/hc-oss/maplibre-gl-styles/master/styles/osm-mapnik/v8/default.json) that we use to define our MapStyle:

https://github.com/meshtastic/web/blob/5fdd07ee68b14c687c71e785b517bdde18f2f910/src/pages/Map.tsx#L115

Because the map is a raster it would be difficult to change the look of it, and OpenStreetMap does not offer a dark mode for the map.

Hunter275, 2024-06-07:
#33 could possibly solve this

Talie5in, 2024-06-07:
Surely we can pull from other maplibre-gl-styles ?

Eg,
https://github.com/teamapps-org/maplibre-gl-styles/blob/main/dark/dark_matter.json
https://github.com/teamapps-org/maplibre-gl-styles/blob/main/dark/dark_theme.json

Hunter275, 2024-06-07:
maplibre-gl just styles maps it does not source the map itself.

The source of the map for those two styles is: 
```
    },
    "openmaptiles:version": "3.x",
    "openmaptiles:mapbox:owner": "openmaptiles",
    "openmaptiles:mapbox:source:url": "mapbox://openmaptiles.4qljc88t",
    "maputnik:renderer": "mbgljs"
  },
  "sources": {
    "openmaptiles": {
      "type": "vector",
      "url": "mbtiles://{v3}"
    }
  },
```

Meaning the map tiles come from [Mapbox](https://www.mapbox.com/), which is a paid service. Mapbox is providing their map as a vector which allows for creating custom layers:
```
{
      "id": "water",
      "type": "fill",
      "source": "openmaptiles",
      "source-layer": "water",
      "filter": ["==", "$type", "Polygon"],
      "layout": {"visibility": "visible"},
      "paint": {"fill-color": "rgba(7, 43, 76, 1)", "fill-antialias": false}
    },
```

OpenStreetMap is free but their map comes as a raster image instead of a vector so there is no ability for styling.

KyleMaas, 2024-06-07:
How about adding a CSS `filter: invert(100%)` to the map canvas?  Trying that on my end, the map looks great.

Hunter275, 2024-06-07:
Inverted doesn't look horrible, I agree.

[image]

Hunter275, 2024-06-09:
https://meshmap.net/ uses the same hack on their map just with `invert(1) hue-rotate(180deg) brightness(1.25)`

commit febd62d8 (2024-06-17): fix #199
closed 2024-06-17 by Hunter275