meshtastic/web #222: Six-digit Bluetooth PIN starting with a zero is truncated when viewed again

KyleMaas, 2024-06-07:
To reproduce, do the following:

1. Set Bluetooth PIN set to `Fixed Pin`
2. Set the PIN to a number starting with a `0`, such as `012345`
3. Save settings
4. Reload the interface and reconnect
5. Go look at the PIN.  You should now see it truncated to `12345` instead of `012345`

garthvh, 2024-06-07:
The value used for BLE pin is a number, it can't start with a 0

adokitkat, 2025-03-16:
This shouldn't obviously be the case... PIN is a collection of numbers 0-9, not a whole number. It shouldn't be not able to start with 0.

Hunter275, 2025-03-16:
Issue should now be to prevent users from setting a PIN with a leading 0

garthvh, 2025-03-16:
The BLE pin in the protobuf is a number and can not start with a 0

danditomaso, 2025-03-17:
This was a regression when I added the BLE field validation. Sorry about that it may have been missed. Please add it to the tests for this component to ensure it doesn't revert again.

danditomaso, 2025-03-29:
Closing as this has been fixed in PR #439

closed 2025-03-29 by danditomaso