meshtastic/web #224: Virtual screen and buttons in the web browser

mkgin, 2024-06-09:
Nice that you are proactively looking for suggestions! I hope the project get's support.
I understand the challenges with firmware, hardware and other resources.

Are you focusing on E-ink or anything connected to new UI features welcome?

I'll try not to spam you to much with every idea and just edit below here for now.

* Virtual screen and buttons in the web browser when connected to the device via the web client .
*

_Originally posted by @mkgin in https://github.com/meshtastic/firmware/discussions/3908#discussioncomment-9691672_

todd-herbert, 2024-06-09:
Hey, sorry, I just want to clarify a bit:

>  * web side mirroring the device screen like a frame buffer or simulating a display controller
> * buttons that work on the device, maybe even acknowledge when clicked and when the firmware has recieved the click.

This specific idea is not really realistic.

I was thinking about it in a different way: what is the reason you'd like to see the device's screen in the Web-UI?
If you were hoping to see a "simplified slideshow of info", similar to the display, I was suggesting that maybe something like this could be added to the UI as its own feature, separate from the display.

If you had a different reason for wanting to mirror the device screen, I'm sorry for leading you off-track.

mkgin, 2024-06-09:
I guess our ideas weren't exactly in sync. Most of the information in the carousel is already available in the web client.

If the UI won't communicate outside the device, I guess this is completely out of scope.

It's pretty impressive what can be done in the browser side.  eg, HD44780 emulator in javascript. So some things seem like  a natural extension. (but not necessarily trivial). Maybe impractical here.

I'll leave this open in case some one from web has any insights. But no big deal if this is closed without comment.

Hunter275, 2024-06-09:
I'm guessing this feature would work like qFlipper, where the app displays whats on the device screen at the same time:
https://github.com/flipperdevices/qFlipper

[image]

todd-herbert, 2024-06-09:
I don't want to straight dismiss the idea. It's not impossible, but as you put it: maybe not trivial.

Right now, the UI is really a few different wrappers, all pretending to be [this one OLED library](https://github.com/ThingPulse/esp8266-oled-ssd1306). The image buffer there does seem to be accessible, so that's one piece of the puzzle.

I imagine the data would be transferred between the web client and the firmware as a set of "admin messages"(?). The web client could send a `get_display_frame_request` message, which would contain the current state of the buttons / controls. The firmware would respond to this with a `get_display_frame_response` message, which would contain the current display frame (a raw 1-bit image), as well as metadata (dimensions, possibly flip-state and rotation?).

One challenge I can see is that displays come in different sizes. I'm not certain about this, but I believe that the size of admin messages are fixed. It might be necessary to use quite a large message, even for small displays?

Another challenge I see is that Meshtastic devices can have quite a large variety of different input sources (buttons, rotary encoders, keyboards, touchscreen etc.). I wonder about how the web client would emulate them, but also about how that information would be encoded for transferring to the node, and how it would be fed in through the firmware code for all the different input sources.

This is all just considering the current UI code; I don't know how the new "work-in-progress" TFT UI would fit in.
  
Have you ever used the [Meshtastic Discord server](https://discord.com/invite/ktMAKGBnBs)? There's a ["moonshoots"](https://discord.com/channels/867578229534359593/1217252717844500480) channel, which is there for people to suggest and discuss big-picture ideas.
> This is a channel for discussing and planning (potential) revolutionary changes to the Meshtastic project. We're looking for ideas that will move the project in an exciting new direction! 🚀 
>
> If you don't see an idea similar to yours, please create a new topic so we can keep discussions focused 😁 
> 
> While there are some limits that we don't want to push (legal, for example), we're looking for ideas that think outside the box. Think of this channel as a place to sell the community on your vision.

It might be a good space to get technical feedback on how to achieve the idea, but also to gauge the opinion of the admin team. That part is definitely a key step in moving forward with anything like this.

mkgin, 2024-06-09:
I'll close this... this is clearly more connected to Display, UI... Understanding a bit more about the implementation, it best to consider this when the state of the New UI development is more clear and what interfaces the new UI code will provide.

I agreed with @todd-herbert. For those interested, moonshots on discord:

> Have you ever used the [Meshtastic Discord server](https://discord.com/invite/ktMAKGBnBs)? There's a ["moonshoots"](https://discord.com/channels/867578229534359593/1217252717844500480) channel

closed 2024-06-09 by mkgin