meshtastic/web #241: [Feature Request]: Autoconnect to Device Hosted Page (ESP32/Portduino)

Talie5in, 2024-06-17:
### Description

Detect on initial page load if the HOST field in the Header responds to the API endpoint and bypass the connection dialog.

This would mean less clicks to get started when using the inbuilt hosted Web client on ESP32 devices or Portduino.

As mentioned in Discord by @Enjayhch 
https://discord.com/channels/867578229534359593/972586878337433670/1252223911135940738

Hunter275, 2024-06-17:
[image]

FamousL, 2024-11-18:
I was actually about to suggest something similar, It seems a touch cumbersome to go through the connection dialogue when there is a serial connection locally that would be useful, even if I could specify a http endpoint as part of the configuration, kind of a way to make the page statically connect to a given endpoint.

kquinsland, 2025-01-25:
Or even storing connections in local storage so the `n` number of IP/BT/serial(?) connections you've had in the past are remembered.