meshtastic/web #251: [Bug]: Saving MQTT config via web interface on Heltec v3 doesn't work

Calimerorulez, 2024-06-23:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.1

### Description

Trying to save MQTT config via web interface doesn't work.

### Relevant log output

```Shell
Console output Chrome:

index-nsbd6Oaw.js:84 Uncaught (in promise) TypeError: o.toBinary is not a function
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
    at Object.writeMessage (index-nsbd6Oaw.js:84:11921)
    at i.toBinary (index-nsbd6Oaw.js:83:29735)
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
    at Object.writeMessage (index-nsbd6Oaw.js:84:11921)
    at i.toBinary (index-nsbd6Oaw.js:83:29735)
    at zg (index-nsbd6Oaw.js:84:14940)
    at Vg (index-nsbd6Oaw.js:84:14228)
```

Hunter275, 2024-06-23:
@dmanierre @Obliterous @kdkd @yaleman @Calimerorulez 

Can you please try this build: https://web-asz33wey0-meshtastic.vercel.app/

MidlifeRenaissanceMan, 2024-06-23:
Seems to have worked on a V3 433MHz board running a5a37cd using an old Chrome Version 103.0.5060.53 (Official Build) (64-bit) on CentOS7. I seem to have some other problems, but I'll try it on a few different V3 units I have

Hunter275, 2024-06-23:
@MidlifeRenaissanceMan other problems?

MidlifeRenaissanceMan, 2024-06-23:
As I suspected. Antenna issues. Not software

PR #252 (2024-06-26): Fix MQTT settings
Fix #251