meshtastic/web #256: [Bug]: Channel > Downlink Enabled does not work

karamo, 2024-06-28:
### Hardware

Not Applicable

### Connection Type

Serial

### Local or Hosted

https://client.meshtastic.org

### Firmware Version

2.3.14

### Description

Channel > Downlink Enable does not work.

Hunter275, 2024-06-28:
I cannot replicate this. I am able to disable/enable Downlink settings and I confirmed they match when paired to an iOS device as well.

closed 2024-06-28 by Hunter275

karamo, 2024-06-28:
Yes, in Web-Client you can enable/disable. > Submit
You are right, ok, now it ist working. I don't know what it was.

Hunter275, 2024-06-28:
@karamo sometimes if you're fast enough you can get to the setting before it loads and any of those changes won't save. That's a bug in itself for the entire app though.