meshtastic/web #259: Channels URL for QR Codes Aren't Properly Formatted

AddisonTustin, 2024-07-06:
### Firmware Version

latest

### What did you do?

The syntax of a URI/URL requires that the fragment (the portion of the URI that follows a `#`) should be at the very end of the URI. When beginning to work on [meshtastic/Meshtastic-Android#953](https://github.com/meshtastic/Meshtastic-Android/issues/953) I noticed that since `?add=true` followed the fragment, Java's native URI feature does not properly parse `add=true` as a query param, and instead includes it as part of the fragment.

There's some cleanup I'd like to do as a part of [meshtastic/Meshtastic-Android#953](https://github.com/meshtastic/Meshtastic-Android/issues/953) to use the URI directly rather than doing manual string processing. So this becomes a small blocker.

### Expected Behavior

The query params (`?add=true`) should be included in the URI before the fragment.

Link to [URI Generic Syntax](https://www.loc.gov/standards/uri/generic.html) (Library of Congress).

### Current Behavior

[When constructing the channelsUrl](https://github.com/meshtastic/web/blob/4653656420d3cc0e7e4fbd36f3f396cbc2b9b156/src/components/Dialog/QRDialog.tsx#L54) `?add=true` is appended to the end of the URI.

### Participation

- [X] I am willing to submit a pull request for this issue.

Hunter275, 2024-07-07:
@AddisonTustin you marked that you were willing to do a PR for this, is that in the works?

AddisonTustin, 2024-07-07:
Haven't started the PR yet. I filed a [similar issue in Meshtastic-Apple](https://github.com/meshtastic/Meshtastic-Apple/issues/752) and want to make sure that there's agreement across the different usages before putting out any changes.

Hunter275, 2024-07-19:
@AddisonTustin where did this land? I believe the iOS app has been updated?

AddisonTustin, 2024-07-19:
I think I can probably submit the pr draft I had. Though it might need to be updated slightly to handle extracting the channel config from old/legacy URLs? 

If that's not a concern then the pr should be good for review.

Hunter275, 2024-07-20:
> I think I can probably submit the pr draft I had. Though it might need to be updated slightly to handle extracting the channel config from old/legacy URLs?
> 
> If that's not a concern then the pr should be good for review.

I don't think we need to worry about legacy URLs

AddisonTustin, 2024-07-20:
Cool, marked #260 as ready for review. Lmk if you've got any feedback

PR #260 (2024-07-24): fix: move qrcode URL query-string before fragment
Resolves #259.

Pictures of changes:

**Create QR Code Dialog**:
[image]

**Import QR Code Dialog**:
[image]
closed 2024-07-24 by Hunter275