meshtastic/web #262: [Bug]: Meshtastic Wisblock Rak 4631 Fault

Davidm7cay, 2024-07-08:
### Hardware

Rak4631

### Connection Type

Bluetooth

### Local or Hosted

http://meshtastic.local

### Firmware Version

2.3.9

### Description

Hi, I hope someone can help me, I have got a Rak 4631 that has developed a fault as seen on log that can be seen from website.  I am using 2.3.9 firmware version as when I flash latest version of stable or Alpha  I get up on my lcd screen Critical Erroe #3. So if I put 2.3.9 on, it all works apart from a few things  that are the  errors you see in log , it wont pick up any other of my nodes, also my other nodes wont pick up my rack node, it seems so unresponsive/slow  to commands from my mobile phone. if you need any more info please feel free to ask   but I do want to know what these mean ERROR | ??:??:?? 711 [RadioIf] ignoring received packet due to error=-7
ERROR | ??:??:?? 907 [RadioIf] ignoring received packet due to error=-7

Thank you 

### Relevant console output

```Shell
Please view https://pastebin.com/yLcd9Sfr for all logs and errors from Firmware Version Alpha 2.3.9
```

Hunter275, 2024-07-08:
Not a web issue, I'd recommend asking in our Discord for help: https://discord.com/invite/ktMAKGBnBs

closed 2024-07-08 by Hunter275

Davidm7cay, 2024-07-08:
I Have asked there but for some reason i have stopped getting help