meshtastic/web #279: Update GitHub Actions

Hunter275, 2024-08-26:
We need to update the GitHub actions now that the repo is getting more active, we shouldn't be publishing PRs and the likes.

TheCyberRonin, 2024-08-30:
May I suggest that you change the scope of the `CI` workflow? You should be able to target a merge into `master` branch. That way anything that actually gets added would be considered "a release". I'd also maybe add a `develop` branch that is a "container" for work that isn't quite ready for a release and be a staging area for a release.

TheCyberRonin, 2024-08-30:
#287 Should address the `CI` workflow (building and doing a release) running on PRs. After that PR is accepted/merged, it will be in effect.

After that is merged, the `CI` GitHub Action will only run when a push to `master` is made. (A PR merge counts as a push)

closed 2025-02-26 by Hunter275