meshtastic/web #282: [Feature Request]: New User Experience

Hunter275, 2024-08-29:
### Description

It can be very overwhelming to load up the web console the first time if you've not used a phone app to setup the device before hand. If a "factory new" device is connected, prompt the user for the bare minimum settings required to get their radio working.

- [ ] Prompt the user for bare minimum configuration settings
  - [ ] Only prompt if the device has not been configured ("factory new")
  - [ ] Only prompt the user one time (how will we remember this?)
  - [ ] Allow user to skip this prompt entirely and go directly to web app
- [ ] Region setting included in prompt
- [ ] Device short name setting included in prompt
- [ ] Device long name setting included in prompt
- [ ] Prompt to download PKI keys (see #281)

powersjcb, 2024-09-13:
Might be nice to have a group brainstorming session on this involving devs from the iOS + Android dev teams.

Having a consistent user experience for onboarding and setup would make it easier for friends to help each-other get started on a new device.

danditomaso, 2025-01-06:
@Hunter275 Any chance we can get this back on track? The last comment on this issue is from Sept 13th, and I think this is valuable piece of the Meshtastic app puzzle we're missing currently. @powersjcb

Hunter275, 2025-01-07:
> @Hunter275 Any chance we can get this back on track? The last comment on this issue is from Sept 13th, and I think this is valuable piece of the Meshtastic app puzzle we're missing currently. @powersjcb

This does need added!

I'm going to focus on getting the long lived PRs reviewed so I won't be of much use, but feel free to get started on this yourself.

danditomaso, 2025-01-07:
@fifieldt Any advice on how to go about finding a active IOS & Android contributor dev, I'd love to have all of us in a Discord voice channel to discuss what this might look like?